witnessmenow/ESP32-Trinity #1: +5v position <>  before Diode or after Diode

Kamal-Sonani, 2021-09-21:
what if I give reverse polarity to board? as per design I think +5v firstly pass through Diode(D2) 

& if not ,Capacitor will Blast but rest of circuit remain.
Correct me 🧐 with logic.

waiting for final release.

witnessmenow, 2021-09-21:
Potentially, the purpose of D2 is to allow the barrel jack to power the esp32 side of the circuit, while providing a stop for the USB from powering the matrix panel unless the user specifically configures for that using the jumper pins.

If you have a project that uses 2A on the panel and you want to program it with USB, there needs to be a way to power the esp32 while not powering the panel.

There is no reverse polarity protection in this version of the design.

witnessmenow, 2021-09-22:
I'm going to close this as I don't feel like there is any action to be taken on it at the moment

closed 2021-09-22 by witnessmenow