witnessmenow/ESP32-Trinity #4: Some Matrix panels when supplied with more than a 5V supply require level shifting.

witnessmenow, 2021-09-25:
This seems to be rare as I have only seen it on one panel out of the 10+ I have tested and I never seen any complaints about the I2S matrix shield (which is effectively the same as the Trinity) having this problem but during testing of the Trinity Makerfabs found a 64x64 panel that had issues when using a power supply that provided more than 5V

Even if the power supply is rated for 5V, some supplies, especially cheaper ones, will combat against voltage drop by providing more than 5V. 

For most displays this is not an issue, I have tested displays here all the way up to 5.6V without any issue, but this one display that Makerfabs had (and now I have) seems to deteriorate if the supply voltage is larger than roughly 5.2V. The display works fine with lower voltages than this so presumably it is not actually a defective panel.

For this problem panel USB-C seems to work well with them as USB goes through the reverse feedback protection diode which will bring down the voltage below the problem voltage range.

To combat these panels being a problem in the future, the next revision of the Trinity will include level shifters from 3.3V of the ESP32 to whatever VCC is of the Matrix panel. I have tested using two 74AHCT541ABQX for the level shifting (as they are compact) and they seem to be working well

[image]

I can make the design available if people want them, but they are kind of difficult to assemble (the level shifters can't be hand soldered)

One challenge of adding the level shifters is it will be no longer possible to use a 2 layer board for the Trinity and we will have to move to a 4 layer, which is something new to me. I have the majority of a 4 layer design made up, but I have not even ordered a trial yet never mind being in a position to swap over to this version for production.

[image]

lols, 2022-09-27:
You can try to use 0ohm resistor to jump pass trace to continue to use 2 layer pcb, not pretty, but gets the job done and is cheaper than 4 layer
But as usual try not to jump unless really no way to route pass a trace

witnessmenow, 2024-02-03:
Added files for the V2 Trinity (which are **untested!**) and the external level shifter to the Hardware section

mariopepe, 2026-01-04:
> witnessmenow
>  left a comment
> (witnessmenow/ESP32-Trinity#4)
>  <https://github.com/witnessmenow/ESP32-Trinity/issues/4#issuecomment-3707502574>No I never did actually, maybe someday!
>
> I tested the concept of it by building a board with the level shifters that
> attached to a V1.x Trinity, but I never assembled a v2 PCB

Thanks so much again!!!! I will 100% let you know if I end up “making” some v2!