witnessmenow/ESP32-Trinity #11: DisplayImageFromWeb example doesn't work with JPEGDEC 1.2.2+

joelhaasnoot, 2021-12-12:
Tested to work with 
```
1.1.1
1.2.0
1.2.1
```

Error with 1.2.2+ is
```
invalid conversion from 'void (*)(JPEGDRAW*)' {aka 'void (*)(jpeg_draw_tag*)'} to 'int (*)(JPEGDRAW*)' {aka 'int (*)(jpeg_draw_tag*)'} [-fpermissive]
```

brianlich, 2022-12-21:
I was able to get around the error above by modifying the following in code.
```
int JPEGDraw(JPEGDRAW *pDraw)
{
  // Stop further decoding as image is running off bottom of screen
  if (  pDraw->y >= dma_display->height() ){
    return 0;
  }

  dma_display->drawRGBBitmap(pDraw->x, pDraw->y, pDraw->pPixels, pDraw->iWidth, pDraw->iHeight);

  return 1;
}
```

However, I'm getting a new error now that looks be in the JPEGDEC library. Any ideas? I'm using PlatformIO.
```
In file included from src/main.cpp:47:
.pio/libdeps/wemos_d1_mini32/JPEGDEC/src/JPEGDEC.h:216:14: error: 'File' has not been declared
     int open(File &file, JPEG_DRAW_CALLBACK *pfnDraw);
              ^~~~
```

clajarac, 2023-09-06:
did you fix it? same problem here

hpsaturn, 2023-10-04:
I had the same  issue:

> In file included from src/main.cpp:47:
> .pio/libdeps/wemos_d1_mini32/JPEGDEC/src/JPEGDEC.h:216:14: error: 'File' has not been declared
>      int open(File &file, JPEG_DRAW_CALLBACK *pfnDraw);
>               ^~~~

I'm using a TFT_eSPI library, and disabling the `SMOOTH_FONT` in the TFT Setup I fixed it. Maybe could have relation with your issue.