witnessmenow/ESP32-Trinity #13: Colon not blinking in between minutes

donnersm, 2022-01-28:
I tried the sketch for the testis clock with WIFImanager.
I used the original code but I changed the display interface to work on my hardwire with different pin setup.
All is working except the colon. Whenever an animation is active the 2 dots are blinking at 1 sec interval but whenever the animation of a number( or letter) is done, the colon freezes in whatever state it was in last. sometime's it stays black, sometimes it stays white but doesn't blink. Blinking is resumed when next animation starts.

I changed Matrix config to:
```
// -------------------------------------
// -------   Matrix Config   ------
// -------------------------------------

#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 2      // Total number of panels chained one to another
#define A_PIN         23
#define B_PIN         19
#define C_PIN         5
#define D_PIN         17
#define E_PIN         22
#define LAT_PIN       18
#define OE_PIN        15
#define CLK_PIN       16
#define R1_PIN        33
#define B2_PIN        13
#define R2_PIN        14
#define G2_PIN        12

https://user-images.githubusercontent.com/60584210/151605967-a93c3f68-ac49-4284-8ad0-7fcd37576fef.mp4

#define G1_PIN        26
#define B1_PIN        27
// -------------------------------------
```
And I added /changed it near line 373 to fit my hardware:

`// -------------------------------------
// -------   Matrix Config   ------
// -------------------------------------

#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 2      // Total number of panels chained one to another
#define A_PIN         23
#define B_PIN         19
#define C_PIN         5
#define D_PIN         17
#define E_PIN         22
#define LAT_PIN       18
#define OE_PIN        15
#define CLK_PIN       16
#define R1_PIN        33
#define B2_PIN        13
#define R2_PIN        14
#define G2_PIN        12
#define G1_PIN        26
#define B1_PIN        27
// -------------------------------------`

witnessmenow, 2022-01-28:
Thanks for the video, it must be a problem with the code for the 12 hour
clock cause the 24 hour clock works fine.

Will dig into over the weekend!

donnersm, 2022-01-28:
I tried the 24 hour clock by adjusting the config in the wifi manager....result is the same. The 24 hour clock has same effect. colon not blinking.
I can make a video but it looks just the same except it's a 24hour clock lol

witnessmenow, 2022-01-29:
What code version is it? Is this the version with wifi manager?

donnersm, 2022-01-29:
Yes it's the version with wifi manger. I used the link you send me on discord and clicked the wifi manager version

donnersm, 2022-01-30:
Hi brian,
I just uploaded the non wifimanager version and it has the same behaviour

donnersm, 2022-01-30:
Found a workaround....is not optimal but it might help you diagnose.
I included the fastled libary:
#include <FastLED_NeoMatrix.h>
as a quick and dirty import of one specific function:
EVERY_N_MILLISECONDS(1000)

So I added this line before the closing bracket of the loop
EVERY_N_MILLISECONDS(1000) {handleColonAfterAnimation();};

Now all is working fine.
For some reason the     
if (finishedAnimating) {
      handleColonAfterAnimation();
    }
loop doesn't do the handleColonAfterAnimation(); the loop itself is entered ( i added a line of text output to serial port to monitor)

Hope that helps

donnersm, 2022-04-02:
Did some more digging ....this only happens with the doublebuffer on ( display)  looks like when it happens, it writes to the other buffer...anyway...due to memory limitations i stick to a single buffer for now. With double buffer disables..all workes fine

ddavehauser, 2022-11-12:
Hi, im currently having the same issue as well.

are there any fixes except the fastled workaround yet?

Thanks!