witnessmenow/ESP32-Trinity #21: 3.3V buffering capacitor needed

jhbruhn, 2023-04-09:
On my Trinity board I had to add an extra capacitor to the 3.3V power line for it to have a stable WiFi connection at all times. I used a 470uF one as it was the first one from the drawer, but less is probably also enough.

Maybe I accidentally used some slightly different parts to assemble my board because it apparently wasn't needed for the "originals", but I'm leaving this info here if anyone should have intermittent WiFi issues like I had.

Thanks for this great project!

jhbruhn, 2023-04-09:
Huh, it seems that does not fix my issues completely. I loose my MQTT connection after some minutes. Touching the ESP32-Casing (???) seems to restore it. I am confusion.

witnessmenow, 2023-04-09:
I haven't seen any similar issues with any of my versions, but happy to keep it open.

Without knowing much about anything, touching the esp32 casing solving it sounds like a ground issue, maybe the ground pads under the esp32 didn't flow right?

jhbruhn, 2023-04-09:
That was my thought as well, but the datasheet states that the bottom pad does not necessarily have to be soldered. 

I switched to a different WiFi AP (was tplink EAP 653, now a FritzBox) and it seems to be stable now, so I think I can blame the WiFi network for my issues. I will need to watch it further but am pretty certain that this was the actual issue.